steve3d/arduino-vscode #4: "compileOptions" are being added after "ino" file

guatedude2, 2017-02-03:
When specifying the "compileOptions" config, they seem to get added after the "ino" file, which the `audrino-builder` tool does not take into account.

guatedude2, 2017-02-03:
Added a PR for the fix
https://github.com/steve3d/arduino-vscode/pull/3

steve3d, 2017-02-04:
your PR has some potential problems, so I've updated the extension to 0.1.4, which will make sure the file part of flag is the last one.  And it also has the real verbose output.

guatedude2, 2017-02-04:
I ended up not being able to use this extension because I have a NodeMCU which uses it's own upload tool, such as `esptool`

I ended up writing these scripts:

*Build Script*
```
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/guatedude/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/guatedude/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Alex/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=160,UploadSpeed=115200,FlashSize=4M3M -ide-version=10801 -build-path /Users/guatedude/Documents/Arduino/ArduinoTest/.build -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Alex/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=/Users/Alex/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -prefs=runtime.tools.mkspiffs.path=/Users/Alex/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 /Users/guatedude/Documents/Arduino/ArduinoTest/ArduinoTest.ino
```

*Upload Script*
```
/Users/guatedude/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -cd nodemcu -cb 115200 -cp /dev/cu.SLAB_USBtoUART -ca 0x00000 -cf /Users/guatedude/Documents/Arduino/ArduinoTest/.build/myArduinoExample.ino.bin
```

It would be nice to support other Arduino devices

steve3d, 2017-02-05:
maybe I can do it with a alternative way, but I can not make any promise on this, because I have no NodeMCU board to test.

steve3d, 2017-02-05:
Looks like this only need some extra compile flags, but I'm not quite understand that why my Arduino IDE creates `myArduinoExample.ino.hex` while yours create the `myArduinoExample.ino.bin` file?

guatedude2, 2017-02-05:
I believe thats the BIN file that gets uploaded via esptool (https://github.com/espressif/esptool)

Alejandro Gonzalez

steve3d, 2017-02-06:
done, please help me test and give some feedback. thanks

guatedude2, 2017-02-06:
sure can

guatedude2, 2017-02-06:
Works like a charm

closed 2017-02-06 by guatedude2

guatedude2, 2017-02-06:
Thanks for the support Steve