steve3d/arduino-vscode #6: NodeMCU / ESP8266 support

ivaylopivanov, 2017-02-05:
As mentioned [here](https://github.com/steve3d/arduino-vscode/issues/4#issuecomment-277454289) it will be nice to have support for other Arduino devices.

steve3d, 2017-02-05:
sorry, I'm not a master of arduino, and this project is just my hobby project, and also this is very first time I heard of this NodeMCU device.

and I don't have this type of device by hand, so even if I want to support it, I can not do that without a actual device.

So sorry for that.

steve3d, 2017-02-06:
done, please help me test and give some feedback. thanks

guatedude2, 2017-02-06:
This one has a couple of problems. One thing I noted that after running builds on mac for several times, the `tmp/sketch` gets populated with tmp redundantly so you get
`tmp/sketch/tmp/sketch/...`, so in my script I use `.build/` because `.` files don't get copied over to the sketch

guatedude2, 2017-02-06:
I think this ticket can be closed

steve3d, 2017-02-06:
You might need to clean or rebuild.

Close this.

closed 2017-02-06 by steve3d