steve3d/arduino-vscode #7: When building "file name too long"

guatedude2, 2017-02-06:
When buliding too many times the sketch gets copied into the `tmp` folder. It's recommended to use something like `.build` which any `.` files get ignored at least in UNIX, I'm not sure in Windows.
```
============== Begin to compile. ==============
open /Users/guatedude/Documents/Arduino/iLifeController/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/sketch/tmp/iLifeController/preproc/ctags_target_for_gcc_minus_e.cpp: file name too long
```

guatedude2, 2017-02-07:
https://github.com/steve3d/arduino-vscode/pull/8

steve3d, 2017-02-14:
merged, thanks

closed 2017-02-14 by steve3d

russellblog, 2017-04-26:
Hi, for problems concerning path too long issues, I suggest you to try the new long path tool. This can help you with all kinds of path too long cases,