steve3d/arduino-vscode #11: avr-g++: error: missing device or architecture after '-mmcu='

khgm, 2017-02-13:
Hi,
[arduino-avr-mega.txt](https://github.com/steve3d/arduino-vscode/files/771868/arduino-avr-mega.txt)

i got that message when i try to compile with fqbn ardunio:avr:mega parameter. I spent several hours to debug where can be the problem without success. Attached the detailed output and parameters.

Thanks for the advice/idea where to dig deeper
Gabor

steve3d, 2017-02-14:
first, please make sure that you are using arduino ide 1.8+
then you will need to find the missing parameter from Arduino IDE by enabling the verbose output.

steve3d, 2017-02-14:
and I think for your board, the correct `fqbn` should be set as `arduino:avr:mega:cpu=atmega2560`

closed 2017-02-14 by steve3d

khgm, 2017-02-14:
that fqbn worked, thanks