steve3d/arduino-vscode #12: Include file not found in include path on Windows

NonsenseInc, 2017-02-15:
I tried building some of the examples supplied with the Arduino IDE. While I can successfully build simple programs like the Blink.ino, programs that require includes do not work with IntelliSense.

For example the DigitalPotControl.ino (also from the Ardiono IDE examples) does compile, but IntelliSense throws:
`file: 'file:///c%3A/Users/Jan/Desktop/DigitalPotControl/DigitalPotControl.ino'
severity: 'Info'
message: 'Include file not found in include path.'
at: '31,1'
source: ''`

My settings are (replacing / with \\\\ doesn't help):
`"arduino.idePath": "C:/Program Files (x86)/Arduino",
"arduino.libraryPath": "C:/Program Files (x86)/Arduino/libraries",
"arduino.serialPort": "COM9"`

How can I help fixing that?

khgm, 2017-02-15:
Good question, i use absolute path for the .h files but i know that is not the right and scalable one. Theoretically the settings in c_cpp_properties.json would be the right one but for me not working.  https://blogs.msdn.microsoft.com/vcblog/2016/03/31/cc-extension-for-visual-studio-code/

NonsenseInc, 2017-02-15:
Yes, I thought about using the c_cpp_properties.json as a workaround and we could certainly get that working. Thank you for the nice idea! 
But then I could also setup the Arduino toolchain as a generic build tool and not use this extension at all. It would be a lot better if the extension does the whole job.

steve3d, 2017-02-17:
@NonsenseInc , this extension is just a simple wrapper for arduino ide, not a replacement.

It depends on the cpp-tools, without it, vscode works like a notepad.

and your libraryPath is wrong, libraryPath should be set to your library, which is the folder when you download the extra library in Arduino IDE, under windows, this should be the `Arudino` folder inside `My Documents`.

steve3d, 2017-02-20:
No feedback, so closing this.

closed 2017-02-20 by steve3d

NonsenseInc, 2017-02-20:
Oh I'm sorry I, I had a lot of work keeping me from browsing here. As I wrote, I am able to compile successfully, so of course i have the Arduino IDE and, as IntelliSense complains, I must have cpptools as well.
If I have to set libraryPath to a single library, how can I then deal with multiple libraries?

steve3d, 2017-02-21:
No matter what platform you use, the Arduino IDE will always download the libraries into sub folder of `Arduino/libraries` of your `Documents` on mac and `My Documents` folder on windows. so under mac, the downloaded libraries are stored under  `/Users/steve/Documents/Arduino/libraries/`

for example if you downloaded the WiFi101 library, the files will be saved to `/Users/steve/Documents/Arduino/libraries/WiFi101` folder, so you just specify the `/Users/steve/Documents/Arduino` as your libraryPath, then it should work.

of course, you can add extra folders to `includePath` of `c_cpp_properties.json` file.

And if this is the case, I don't think this is a _issue_