steve3d/arduino-vscode #18: Unknown package error

tejas007tez, 2017-03-31:
I don't think it's an issue. Since there was no gitter chat option available, I am posting this as an issue.

I just wanted to ask, I have configured these settings:

```"arduino.idePath": "E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/",
    "arduino.libraryPath": "E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/",
    "arduino.uploader": "E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/Portable/packages/esp8266/tools/esptool/0.4.9/",
    "arduino.fqbn": "esp8266:esp8266:nodemcu:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M",
    "arduino.verbose": true,
    "arduino.serialPort": "COMx",
    
    "arduino.uploadOptions": "-vv -cd ck -cb 115200 -cp COMx -ca 0x00000 -cf C:/Users/Tejas/AppData/Local/Temp/arduino_build_990450/NINJA_Server.ino.bin"```

Are these correct? Any additional lines to be added still?

steve3d, 2017-04-01:
if you write that kind of uploadOptions, then the extension will always use the specified file to upload no matter what this the real source file.

why not use these?
```
"arduino.uploadOptions": "-vv -cd ck -cb $BAUDRATE -cp $SERIALPORT -ca 0x00000 -cf $TARGET.bin",
```

steve3d, 2017-04-01:
please check the README.md, the example section are set for esp8266 compile and upload

tejas007tez, 2017-04-01:
Thank you for the response. Now I have changed specific values to variables as you suggested. But the "Unknown package" error persists.
And yes, I have set the settings similar to the example that you have showed.

steve3d, 2017-04-10:
for esp8266, you will need to set custom `compilerOptions`, and you `uploader` should point to the  uploader program, not uploader path.

check README.md again.

CodingCreate101, 2017-04-10:
I am sorry. That was my mistake. I didn't configure ```compilerOption```, there was a season, though.

Anyway, now **Build** is working fine after making some-more changes. But not able to **Upload**.

Error:

```
============== Begin to upload. ==============
E:\iThings\arduino-1.8.1-windows\arduino-1.8.1\hardware\tools\avr\bin\avrdude.exe -CE:\iThings\arduino-1.8.1-windows\arduino-1.8.1\\hardware\tools\avr\etc\avrdude.conf -patmega328p -carduino -PCOMx -b115200 -vv -cd ck -cb $BAUDRATE -cp $SERIALPORT -ca 0x00000 -cf $TARGET.bin -Uflash:w:e:\iThings\My projects\Ninja_new\NINJA_Server\.build\NINJA_Server\NINJA_Server.ino.hex:i

avrdude.exe: Version 6.3, compiled on Dec 16 2016 at 13:33:19
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "E:\iThings\arduino-1.8.1-windows\arduino-1.8.1\\hardware\tools\avr\etc\avrdude.conf"

avrdude.exe: Can't find programmer id "d"

```

Any idea??

steve3d, 2017-04-10:
avrdude can not be used as uploader for esp8266. that's why your upload will never work.

CodingCreate101, 2017-04-10:
So, what I should use instead? I mean, I am able to Upload using regular Arduino IDE.. What setting I should change?

CodingCreate101, 2017-04-10:
Yeah, of course. AVRDude can not be used. There esptool.exe for that. But I have configured it. Then why it's taking avr dude? Where to change that?

steve3d, 2017-04-10:
the four options showed in README.md are exact options needed for esp8266. try those.

CodingCreate101, 2017-04-10:
This is what You have given:
```
"arduino.fqbn": "esp8266:esp8266:nodemcu:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M",
"arduino.uploader" : "/Users/steve/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool",
"arduino.uploadOptions": "-vv -cd ck -cb $BAUDRATE -cp $SERIALPORT -ca 0x00000 -cf $TARGET.bin",
"arduino.compileOptions": "-hardware /Users/steve/Library/Arduino15/packages -tools /Users/steve/Library/Arduino15/packages -prefs=runtime.tools.esptool.path=/Users/steve/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/steve/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/steve/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2"
```

This is what I have configured:
```
"arduino.idePath": "E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/",
    "arduino.libraryPath": "E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/",
    "arduino.uploader": "E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/Portable/packages/esp8266/tools/esptool/0.4.9/esptool.exe",
    "arduino.fqbn": "esp8266:esp8266:nodemcu:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M",
    "arduino.verbose": true,
    "arduino.serialPort": "COMx",
    "arduino.uploadOptions": "-vv -cd ck -cb $BAUDRATE -cp $SERIALPORT -ca 0x00000 -cf $TARGET.bin",
    "arduino.compileOptions": "-hardware E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/Portable/packages -tools E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/Portable/packages -prefs=runtime.tools.esptool.path=E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/Portable/packages/esp8266/tools/esptool/0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/Portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=E:/iThings/arduino-1.8.1-windows/arduino-1.8.1/Portable/packages/esp8266/tools/mkspiffs/0.1.2",
    "files.autoSave": "afterDelay"
```

But there are these defaults:
```
"arduino.partno": "atmega328p" 
"arduino.programmer": "arduino"
```
What I should change those to?